hansonrobotics/HEAD #1: blender not connected

mjsduncan, 2016-05-09:
after installing and building everything with hrtool and running "dev.sh", the blender head is static eventhough according to the ocbtree shell face tracking is working and emotions are being published for display.

this is blender version:
````
biocog@topdog:~/hansonrobotics$ dpkg -l | grep blender
ii  blender                                               2.77-a-1459879952-0thomas~trusty0                   amd64        Very fast and versatile 3D modeller/renderer

````
this is python path:
````
biocog@topdog:~/hansonrobotics$ echo $PYTHONPATH
/opt/ros/indigo/lib/python2.7/dist-packages:/usr/local/lib/python2.7/dist-packages:/home/biocog/opencog/opencog/python:/home/biodata/opencog/opencog/nlp:/home/biodata/opencog/build/opencog/cython:/usr/local/share/opencog/python

````
this is the error message from the blender shell:
````
connect failed: No such file or directory
read blend: /home/biocog/hansonrobotics/HEAD/src/blender_api/Sophia.blend
Dependency cycle detected:
  deform depends on control through Child Of.
  control depends on deform through Locked Track.

Traceback (most recent call last):
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/Sophia.blend/loader.py", line 15, in <module>
    register()
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/Sophia.blend/loader.py", line 13, in register
    rigControl.init()
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/__init__.py", line 47, in init
    animationManager.init()
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/animationManager.py", line 580, in init
    bpy.evaAnimationManager = AnimationManager()
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/animationManager.py", line 76, in __init__
    self.actuatorManager = ActuatorManager()
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/actuators.py", line 83, in __init__
    import artistic.actuators
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/artistic/actuators.py", line 2, in <module>
    import numpy as np
  File "/usr/local/lib/python2.7/dist-packages/numpy/__init__.py", line 180, in <module>
    from . import add_newdocs
  File "/usr/local/lib/python2.7/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/local/lib/python2.7/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/local/lib/python2.7/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/usr/local/lib/python2.7/dist-packages/numpy/core/__init__.py", line 14, in <module>
    from . import multiarray
ImportError: /usr/local/lib/python2.7/dist-packages/numpy/core/multiarray.so: undefined symbol: _Py_ZeroStruct
Command Source 'ros' won't build
No Command Sources found
Traceback (most recent call last):
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/blenderPlayback.py", line 107, in modal
    eva = bpy.evaAnimationManager
AttributeError: module 'bpy' has no attribute 'evaAnimationManager'

location: <unknown location>:-1

location: <unknown location>:-1
Traceback (most recent call last):
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/blenderUI.py", line 78, in draw
    for i, emo in enumerate(bpy.evaAnimationManager.bones):
AttributeError: module 'bpy' has no attribute 'evaAnimationManager'

location: <unknown location>:-1

location: <unknown location>:-1
````

bgoertzel, 2016-05-09:
Does your machine have a GPU?  Is is possible your machine just sucks too
badly for Blender to run such a complex model?

demaris, 2016-05-09:
even with intel builtin or non proprietary drivers you should get 7-14 fps.

I think one time my intern Jonas had this happen and it got fixed by
installing the proprietary drivers instead of default nouveau drivers.

mjsduncan, 2016-05-09:
i have a fancy gpu and i installed the nvidia drivers but maybe they aren't set up correctly?

mjsduncan, 2016-05-09:
during the cogathon broadcast on the hanson "software" slack channel, Vytas Krisciunas thought it was a problem with python2.7 vs python3 paths, that's why i included mine...

wenwei-dev, 2016-05-10:
I'm wondering how did you install blender? Your blender version is higher than what we use.

2.71~git201406121839.169c95b-0irie1~trusty1

demaris, 2016-05-10:
I think Mark runs ok with 2.73,  I tried it once also to see if it solved
some problem (it didn't but it ran).

mjsduncan, 2016-05-10:
i did have deb repository with  the newer version so i removed it and ran hrtools -i, now i have:
````
blender --version
Blender 2.71 (sub 0)
	build date: 2014-06-13
	build time: 08:56:24
	build commit date: 2014-06-12
	build commit time: 18:39
	build hash: 169c95b
	build platform: Linux
	build type: RelWithDebInfo
	build c flags:  -Wall -Wcast-align -Werror=declaration-after-statement -Werror=implicit-function-declaration -Werror=return-type -Wstrict-prototypes -Wmissing-prototypes -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wlogical-op -Wundef -Winit-self -Wnonnull -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wuninitialized -Wredundant-decls -Wno-error=unused-but-set-variable  -fopenmp  -msse2  -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing
	build c++ flags:  -Wredundant-decls -Wall -Wno-invalid-offsetof -Wno-sign-compare -Wlogical-op -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wuninitialized -Wundef -Wmissing-declarations  -D__STDC_CONSTANT_MACROS -fopenmp  -msse2  -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing
	build link flags:  -Wl,--version-script=/build/buildd/blender-2.71~git201406121839.169c95b/source/creator/blender.map -pthread
	build system: CMake
````
with the same git hash but i'm still getting identical error messages on the blender shell.
when i ran "make test" in the opencog repos in atomspace the backward chainer test failed and in opencog the restapi test failed.

mjsduncan, 2016-05-10:
i also get these errors in the ros shell but i didn't think they were relevant since i'm not hooked up to motors...:
````
...
process[joint_state_publisher-18]: started with pid [25631]
process[robot_state_publisher-19]: started with pid [25645]
process[faces_broadcaster-20]: started with pid [25649]
process[state_publisher-21]: started with pid [25655]
process[rviz-22]: started with pid [25657]
Traceback (most recent call last):
  File "/home/biocog/hansonrobotics/HEAD/src/hardware/motors_safety/scripts/safety_node.py", line 3, in <module>
    import rospy
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 47, in <module>
    import yaml
  File "/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg/yaml/__init__.py", line 2, in <module>
ImportError: No module named 'error'
process[performances-23]: started with pid [25667]
process[webui/node_configuration-24]: started with pid [25677]
process[rosbridge_websocket-25]: started with pid [25696]
[robot/motors_safety-4] process has died [pid 25510, exit code 1, cmd /home/biocog/hansonrobotics/HEAD/src/hardware/motors_safety/scripts/safety_node.py __name:=motors_safety __log:=/home/biocog/.hr/log/3ff99392-165c-11e6-8a82-206a8aa0476d/robot-motors_safety-4.log].
log file: /home/biocog/.hr/log/3ff99392-165c-11e6-8a82-206a8aa0476d/robot-motors_safety-4*.log
process[rosapi-26]: started with pid [25704]
[hr.ros_pololu.ros_pololu_node][WARN] [1462849615.765535]: Error creating the motor controller
[hr.ros_pololu.ros_pololu_node][WARN] [1462849615.765701]: could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
Hot keys: 
        q - quit the program
        c - delete current features
        t - toggle text captions on/off
        f - toggle display of features on/off
        n - toggle "night" mode on/off
        a - toggle auto face tracking on/off

/home/biocog/hansonrobotics/HEAD/devel/lib/python2.7/dist-packages/ros2opencv/__init__.py:102: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
````

mjsduncan, 2016-05-12:
maybe this is a problem with python paths?  the error traces back to my local 2.7 installation but i see above there is a ros installation also...

demaris, 2016-05-13:
I've seen and heard of people getting into trouble with virtual
environments, multiple python installs without VEs, and  pre-existing ros
installations.
Is your system a clean ubuntu 14.04 install or HEAD on top of a system with

mjsduncan, 2016-05-15:
i already had 2 versions of ros installed.  can someone post what the $PYTHONPATH looks like for a functioning HEAD installation?

mjsduncan, 2016-05-16:
in the tmux SHELL window this happens:
````
rostopic echo -n 1 /blender_api/available_gestures
ERROR: Cannot load message class for [blender_api_msgs/AvailableGestures]. Are your messages built?
````

wenwei-dev, 2016-05-16:
@mjsduncan It looks like you didn't source the ROS environment variables. That is why you get "Cannot load message" error. Try to run "source ~/hansonrobotics/HEAD/devel/setup.bash" before using rostopic.

mjsduncan, 2016-05-17:
i uninstalled all the ros code and had it reinstalled through hrtool.  now in tmux 6 i get:
````
~/hansonrobotics/HEAD/scripts$ rostopic echo -n 1 /blender_api/available_gestures                         
rostopic: command not found
~/hansonrobotics/HEAD/scripts$ source ~/hansonrobotics/HEAD/devel/setup.bash                    
~/hansonrobotics/HEAD/scripts$ rostopic echo -n 1 /blender_api/available_gestures
````
and then nothing, i have to ctrl-c to get a cursor.

here are the other errors i'm getting in shells:
from tmux 0
````
process[state_publisher-21]: started with pid [3653]
Traceback (most recent call last):
  File "/home/biocog/hansonrobotics/HEAD/src/hardware/motors_safety/scripts/safety_node.py", line 3, in <module>
    import rospy
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 47, in <module>
    import yaml
  File "/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg/yaml/__init__.py", line 2, in <module>
ImportError: No module named 'error'
process[rviz-22]: started with pid [3710]
````
should it be looking in the hanson environment instead of local?

from tmux 2
````
Traceback (most recent call last):
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/blenderUI.py", line 78, in draw
    for i, emo in enumerate(bpy.evaAnimationManager.bones):
AttributeError: 'module' object has no attribute 'evaAnimationManager'
````
from tmux 4
````
Publish expression: confused
[rospy.internal][ERROR] [1463452228.401968]: Unable to initiate TCP/IP socket to topdog:45840 (http://topdog:36435/): Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 619, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 580, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: [/rosbridge_websocket] is not a publisher of [/behavior_switch]. Topics are [['/rosout', 'rosgraph_msgs/Log']]
````
and this is what i get on the web monitor:

[image]

any ideas?

wenwei-dev, 2016-05-17:
> i uninstalled all the ros code and had it reinstalled through hrtool. now
> in tmux 6 i get:
>
> ~/hansonrobotics/HEAD/scripts$ rostopic echo -n 1 /blender_api/available_gestures
> rostopic: command not [...]
This *ImportError: No module named 'error' *looks suspicious. What will
happen if you run "import yaml" in your python interpreter?

> from tmux 2
>
> Traceback (most recent call last):
>   File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/blenderUI.py", line 78, in draw
>     for i, emo in enumerate(bpy.evaAnimationManager.bones):
> AttributeError: 'module' object has no attribute 'evaAnimationManager'
Have you run `./hrtool -b` to build the source?

vytasrgl, 2016-05-17:
Means that python3 cant find ROS in its path. Note that Blender and Motors Safety package use python3.

The following packages need to be installed: rospy, rospkg, blender_api_mgs (from source from the src directory)

Also need to make sure the messages files are accessible for python3. Usually packages would be generated in here, so make sure python3 path has it.
~/hansonrobotics/HEAD/devel/lib/python2.7/dist-packages/

vytasrgl, 2016-05-17:
Also ROS path should be in python3 path as well:
'/opt/ros/indigo/lib/python2.7/dist-packages'

mjsduncan, 2016-05-17:
@wenwei-dev : i've been keeping up to date using ./hrtool -UB regularly.
@vytasrgl : now we're getting somewhere!  i'm a total python noob, how do i install "from source from the src directory", and is the python3 path also $PYTHONPATH ?  should these additions go at the beginning or end?  thanks.

vytasrgl, 2016-05-17:
Installing from src
```
cd ~/hansonrobotics/HEAD/src
sudo pip3 install ./blender_api_msgs/
```
Make sure no other ROS installation beeing sourced on ~/.bashrc file.
Source the HR workspace:

`source ~/hansonrobotics/HEAD/devel/setup.bash`
Check your python path and see if the directories beeing added. They should be added by sourcing workspace

$PYTHONPATH is appended to default python path.; To check full python path:
```
python3
>> import sys
>> print(sys.path)
```

mjsduncan, 2016-05-17:
ok, i did all this and am getting the same thing.  here is `print(sys.path)` in python 3, with paths to HEAD, ros, and my local/lib/python:
````
>>> print(sys.path)
['', '/usr/local/lib/python2.7/dist-packages/CrossMap-0.1.6-py2.7-linux-x86_64.egg', '/usr/local/lib/python2.7/dist-packages/Unidecode-0.04.17-py2.7.egg', '/usr/local/lib/python2.7/dist-packages/openpyxl-1.8.6-py2.7.egg', '/usr/local/lib/python2.7/dist-packages/pandas-0.15.2-py2.7-linux-x86_64.egg', '/usr/local/lib/python2.7/dist-packages/distribute-0.7.3-py2.7.egg', '/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg', '/usr/local/lib/python2.7/dist-packages/docopt-0.6.2-py2.7.egg', '/usr/local/lib/python2.7/dist-packages/bx_python-0.7.2-py2.7-linux-x86_64.egg', '/home/biocog/projects/puzzle', '/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg', '/usr/local/lib/python2.7/dist-packages/docopt-0.6.2-py2.7.egg', '/home/biocog/hansonrobotics/HEAD/devel/lib/python2.7/dist-packages', '/opt/ros/indigo/lib/python2.7/dist-packages', '/usr/local/lib/python2.7/dist-packages', '/home/biocog/opencog/opencog/python', '/home/biocog/opencog/opencog/nlp', '/home/biocog/opencog/build/opencog/cython', '/usr/local/share/opencog/python', '/usr/lib/python3.4', '/usr/lib/python3.4/plat-x86_64-linux-gnu', '/usr/lib/python3.4/lib-dynload', '/usr/local/lib/python3.4/dist-packages', '/usr/lib/python3/dist-packages']
````
i did notice that python3 doesn't recognize my pyyaml module, although python does.
````
biocog@topdog:~/hansonrobotics/HEAD/scripts$ python
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import yaml
>>> yaml
<module 'yaml' from '/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg/yaml/__init__.pyc'>
>>> yaml.error
<module 'yaml.error' from '/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg/yaml/error.pyc'>
>>> quit()
biocog@topdog:~/hansonrobotics/HEAD/scripts$ python3
Python 3.4.3 (default, Oct 14 2015, 20:28:29) 
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import yaml
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg/yaml/__init__.py", line 2, in <module>
ImportError: No module named 'error'
>>> 

````
one other point i haven't mentioned:  the opencog repos in `hansonrobotics/opencog` are symbolic links to my existing repos. could this or the opencog python paths be a problem?

thanks for the help, i have a knack for fucking things up in obscure ways :P

vytasrgl, 2016-05-18:
Did you try:
sudo pip3 install pyyaml

mjsduncan, 2016-05-18:
yes, it says it's already up-to-date!

wenwei-dev, 2016-05-18:
> python3
> "/usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg/yaml/__init__.py",
> line 2, in <module>

It's still looking for pyyaml in python2 path. Can you find yaml in python3
path?

On my machine it's like this
python3
Python 3.4.3 (default, Oct 14 2015, 20:28:29)
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import yaml
>>> yaml
<module 'yaml' from '/usr/lib/python3/dist-packages/yaml/__init__.py'>
>>>

mjsduncan, 2016-05-18:
ok i made some progress by uninstalling pyyaml with pip then installing it with pip3.  tmux 2 now shows activity but continually produces errors similar to this (with the same final line `AttributeError: 'module' object has no attribute 'evaAnimationManager'` as before:
````
 File "/home/biocog/hansonrobotics/HEAD/devel/lib/python2.7/dist-packages/roscom/__init__.py", line 130, in _handle
    self.cmd_func(msg)
  File "/home/biocog/hansonrobotics/HEAD/devel/lib/python2.7/dist-packages/roscom/__init__.py", line 179, in setSomaState
    api.setSomaState(state)
  File "/home/biocog/hansonrobotics/HEAD/src/blender_api/rigControl/commands.py", line 102, in setSomaState
    bpy.evaAnimationManager.setCycle(name=name,
AttributeError: 'module' object has no attribute 'evaAnimationManager'
````
and the web ui shows blender and "safety" are working now:
[image]

wenwei-dev, 2016-05-19:
Please show me if rospkg is installed correctly in python3. Try the
commands as below.

cd ~/hansonrobotics/HEAD
source devel/setup.bash
python3
>>> import rospkg
>>> rospkg
<module 'rospkg' from
'/usr/local/lib/python3.4/dist-packages/rospkg/__init__.py'>

mjsduncan, 2016-05-19:
````
~/hansonrobotics/HEAD$ source devel/setup.bash
~/hansonrobotics/HEAD$ python3
Python 3.4.3 (default, Oct 14 2015, 20:28:29) 
>>> import rospkg
>>> rospkg
<module 'rospkg' from '/usr/local/lib/python3.4/dist-packages/rospkg/__init__.py'>
>>> 
````

mjsduncan, 2016-05-21:
it is a path issue!  if i put `/usr/local/lib/python3.4/dist-packages` in $PYTHONPATH before the python2.7 then blender works fine but face tracking and the web UIs don't work. and the facetracking shell gives a numpy import error.  any suggestions for solving this problem?

mjsduncan, 2016-05-23:
i finally fixed it using a dynamic PYTHONPATH entry as described at [http://stackoverflow.com/questions/19630123/should-pythonpath-have-directories-specific-to-python2-and-python3?rq=1]

closed 2016-05-23 by mjsduncan

linas, 2016-05-24:
I'm not sure how the shell scripts are set up here, but "normally" the blender stuff and the face tracker run in different shells (so that you can track the error messages individually).  And so, normally, you can set different python paths in each shell.  Why that is not being done here .. I don't know.  Perhaps teh shell scripts are buggy?

linas, 2016-05-24:
s/ you can set different python paths in each shell/different paths should be set in each shell/