hansonrobotics/HEAD #2: no ~/.hr/aiml directory after HEAD installation

mjsduncan, 2016-05-16:
concurrently with the blender fail in issue one, in the chatserver shell (tmux window 3) i get this error:
`sh: 1: cd: can't cd to /home/biocog/.hr/aiml/character_dev`
i have a .hr directory with megabytes of log files but no aiml directory.  is that a separate installation?

wenwei-dev, 2016-05-16:
This error can be ignored. This PR #3 should fix this.

mjsduncan, 2016-05-16:
thanks,  new build just has blank screen in window three for dev.sh.  is that correct?

wenwei-dev, 2016-05-26:
Do you still have error in this window? Or is the chatbot working? If it's working, we can close it now.

mjsduncan, 2016-05-26:
ok with current pull i get this in tmux 3:
````
[hr.chatbot.character][INFO] 2016-05-26 10:22:51,198: [generic] Load /home/biocog/hansonrobotics/HEAD/src/chatbot/src/server/characters/../../../aiml/generic/*.aiml
[hr.chatbot.loader][INFO] 2016-05-26 10:22:51,198: Add characters [<Character id: dummy, name: dummy, level: 99>, <Character id: sophia, name: sophia, level: 90>, <Character id: generic, name: generic, level: 93>]
[werkzeug][INFO] 2016-05-26 10:22:51,204:  * Running on http://0.0.0.0:8001/ (Press CTRL+C to quit)
[werkzeug][INFO] 2016-05-26 10:22:51,866: 127.0.0.1 - - [26/May/2016 10:22:51] "GET /v1.1/start_session?botname=sophia&Auth=AAAAB3NzaC&user=hr HTTP/1.1" 200 -
[hr.chatbot.server.response_cache][INFO] 2016-05-26 10:24:53,300: Nothing to dump
[hr.chatbot.server.session][INFO] 2016-05-26 10:24:53,300: Reset session 548256ac-234d-11e6-99f1-206a8aa0476d
[hr.chatbot.server.response_cache][INFO] 2016-05-26 10:26:54,399: Nothing to dump
[hr.chatbot.server.session][INFO] 2016-05-26 10:26:54,399: Reset session 548256ac-234d-11e6-99f1-206a8aa0476d

[hr.chatbot.server.response_cache][INFO] 2016-05-26 10:28:55,505: Nothing to dump
[hr.chatbot.server.session][INFO] 2016-05-26 10:28:55,505: Reset session 548256ac-234d-11e6-99f1-206a8aa0476d
````
then in a new shell if i `telnet localhost 8001` it connects but i get no prompt and any text gets me booted by server:
````
hello
<head>
<title>Error response</title>
</head>
<body>
<h1>Error response</h1>
<p>Error code 400.
<p>Message: Bad request syntax ('hello').
<p>Error code explanation: 400 = Bad request syntax or unsupported method.
</body>
Connection closed by foreign host.
````
and this is what shows up in the chatbot shell:
````
[werkzeug][ERROR] 2016-05-26 10:29:40,307: 127.0.0.1 - - [26/May/2016 10:29:40] code 400, message Bad request syntax ('hello')
[werkzeug][INFO] 2016-05-26 10:29:40,307: 127.0.0.1 - - [26/May/2016 10:29:40] "hello" 400 -
````
so this means something still isn't working, correct?

mjsduncan, 2016-05-26:
since there seems to be a general failure connecting parts as described above,  if this fail looks like part of #14 we can close this issue.

wenwei-dev, 2016-05-27:
Telnet won't work. That is not the way to connect to chatbot server because it's a HTTP server. You can try to use `HEAD/src/chatbot/src/server/client.py` to connect to the server, or use curl to send HTTP request. 

I can see the chatbot server is running normally. So I close this issue.

closed 2016-05-27 by wenwei-dev