hansonrobotics/HEAD #14: blender, chatserver, and OCBTree error.

mjsduncan, 2016-05-26:
current pull builds with no errors.  blender head bobs and blinks, but blender tmux shell has this output:
````
connect failed: No such file or directory
read blend: /home/biocog/hansonrobotics/HEAD/src/blender_api/Sophia.blend
Dependency cycle detected:
  deform depends on control through Child Of.
  control depends on deform through Locked Track.

Command Source 'ros' won't build
No Command Sources found
````
and OCBTree shell crashes:
````
Listening on port 17020
--- Wake up! Elapsed: 1464272569.251680 seconds
----- Behavior pub: Waking up
Publish soma state: normal intensity: 0.1
GNU Guile 2.0.9
Copyright (C) 1995-2013 Free Software Foundation, Inc.

Guile comes with ABSOLUTELY NO WARRANTY; for details type `,show w'.
This program is free software, and you are welcome to redistribute it
under certain conditions; type `,show c' for details.

Enter `,help' for help.
scheme@(guile-user)> Backtrace:
In ice-9/boot-9.scm:
 157: 2 [catch #t #<catch-closure 1802dc0> ...]
In unknown file:
   ?: 1 [apply-smob/1 #<catch-closure 1802dc0>]
   ?: 0 [opencog-extension cog-evaluate! ((DefinedPredicateNode "main loop")
)]

ERROR: In procedure opencog-extension:
ERROR: Throw to key `C++-EXCEPTION' with args `("cog-evaluate!" "Expecting a number, got this: (GetLink\n  (VariableNode \"$int-min\") ; [1449][1]\n  (StateLink\n    (ListLink\n      (ConceptNode \"wake-up\") ; [609][1]\n      (SetLink\n        (ListLink\n          (NumberNode \"0.600000\") ; [475][1]\n          (ConceptNode \"happy\") ; [513][1]\n        ) ; [3838][1]\n        (ListLink\n          (NumberNode \"0.100000\") ; [572][1]\n          (ConceptNode \"irritated\") ; [635][1]\n        ) ; [3839][1]\n        (ListLink\n          (NumberNode \"0.300000\") ; [490][1]\n          (ConceptNode \"surprised\") ; [446][1]\n        ) ; [3840][1]\n      ) ; [3841][1]\n      (SchemaNode \"duration-min\") ; [460][1]\n    ) ; [18446744073709551615][18446744073709551615]\n    (VariableNode \"$int-min\") ; [1449][1]\n  ) ; [18446744073709551615][18446744073709551615]\n) ; [18446744073709551615][18446744073709551615]\n (/home/biocog/hansonrobotics/opencog/atomspace/opencog/atoms/core/RandomNumber.cc:84)")'.

scheme@(guile-user)>
````
chatbot shell produces error described in issue #2 .
face tracking works fine.  i suspect a ros associated python path issue, possibly related to restapi unit test failure in [opencog/opencog](https://github.com/opencog/opencog/issues/2141)

mjsduncan, 2016-05-26:
here's a screen shot showing the webUI monitor page with errors:
[image]

mjsduncan, 2016-05-27:
apparently the chatbot is working, it's just not detected by UI.  same errors in blender and OCBTree for commit 3e5110.

wenwei-dev, 2016-05-27:
Hi @mjsduncan , webui monitoring is going to be updated. 

> Command Source 'ros' won't build
> No Command Sources found

I've fixed a bug in hrtool that used pip2 to install blender_api_msgs. You can pull the latest master and run `./hrtool -b` again. After that you should not see such error.

mjsduncan, 2016-05-27:
thanks @wenwei-dev  no issue in the blender shell now.  i did notice that in the rviz window the eye vectors point up when the the blender head eyes look down and vice versa.  horizontal motion matches.

linas, 2016-05-27:
Is the `"cog-evaluate!" "Expecting a number` thing still happening?  If so, I think I may know the cause:  I made some changes a few days ago that might be causing this.  Apparently, the unit tests didn't catch the issue.

wenwei-dev, 2016-05-27:
@linas The crash of btree was probably because the blender_api_msgs was not found.

mjsduncan, 2016-06-02:
@linas it's still happening for me.
from ocbtree window (tmux 4):
````
scheme@(guile-user)> Backtrace:
In ice-9/boot-9.scm:
 157: 2 [catch #t #<catch-closure 2c02d60> ...]
In unknown file:
   ?: 1 [apply-smob/1 #<catch-closure 2c02d60>]
   ?: 0 [opencog-extension cog-evaluate! ((DefinedPredicateNode "main loop")
)]

ERROR: In procedure opencog-extension:
ERROR: Throw to key `C++-EXCEPTION' with args `("cog-evaluate!" "Expecting a number, got this: (GetLink\n  (VariableNode \"$int-min\") ; [1449][1]\n  (StateLink\n    (ListLink\n      (ConceptNode \"wake-up\") ; [609][1]\n      (SetLink\n        (ListLink\n          (NumberNode \"0.600000\") ; [475][1]\n          (ConceptNode \"happy\") ; [513][1]\n        ) ; [3838][1]\n        (ListLink\n          (NumberNode \"0.100000\") ; [572][1]\n          (ConceptNode \"irritated\") ; [635][1]\n        ) ; [3839][1]\n        (ListLink\n          (NumberNode \"0.300000\") ; [490][1]\n          (ConceptNode \"surprised\") ; [446][1]\n        ) ; [3840][1]\n      ) ; [3841][1]\n      (SchemaNode \"duration-min\") ; [460][1]\n    ) ; [18446744073709551615][18446744073709551615]\n    (VariableNode \"$int-min\") ; [1449][1]\n  ) ; [18446744073709551615][18446744073709551615]\n) ; [18446744073709551615][18446744073709551615]\n (/home/biocog/hansonrobotics/opencog/atomspace/opencog/atoms/core/RandomNumber.cc:84)")'.
````

wenwei-dev, 2016-06-03:
The OpenCog stack used by HEAD is the forked version in hansonrobotics github domain. It's not updated very often. The last catch up was 22 days ago.

wenwei-dev, 2016-06-03:
The forked version works fine with the robot. But maybe we want to explore the latest OpenCog stack, then my solution of balancing the need of stable and new is using different hrtool.

bgoertzel, 2016-06-03:
Adding an option to hrtool would work...

wenwei-dev, 2016-06-03:
I've added an option -d to switch to latest OpenCog stack. 

`./hrtool -d`

mjsduncan, 2016-06-03:
i've been using the current stack, i linked the HEAD opencog directories to my original opencog repos.

mjsduncan, 2016-06-07:
ok the behavior tree doesn't crash and the robot tracks my face so i'm going to close this.

closed 2016-06-07 by mjsduncan