hansonrobotics/HEAD #63: Error when you install vision dependencies

N-zewge, 2016-07-08:
md5sum: openface/celeb-classifier.nn4.small2.v1.pkl: No such file or directory
/opt/hansonrobotics/vision/openface/models/get-models.sh: line 74: [: 199a2c0d32fd0f22f14ad2d248280475: unary operator expected
+ ERROR! openface/celeb-classifier.nn4.small2.v1.pkl md5sum did not match.
  + Expected: 199a2c0d32fd0f22f14ad2d248280475
  + Actual: 
  + Please manually delete this file and try re-running this script.

tesYolan, 2016-07-08:
@N-zewge Delete the /opt/hansonrobotics/openface/ folder and rerun ./hrtool -i

wenwei-dev, 2016-07-08:
This may happen when the upstream openface repository has been updated and/or the model. In order to remove this uncertainty, I think it's better to fork it to hansonrobotics, and add the corresponding model. Does it make sense?

tesYolan, 2016-07-08:
@wenwei-dev The error is most like to be a data corruption. As those models are gotten using the get_model.sh script in the openface models directory. May be to by pass those steps we could use wget commands to locations we own; but i don't think bypassing the checksum would have a good effect especially on the nn4.small, shape_predictor.dat.  as the code is dependent on those particular things doing there job.

wenwei-dev, 2016-07-08:
The error is because the model file openface/celeb-classifier.nn4.small2.v1.pkl was not downloaded. It's actually the bug of the get_model.sh as I looked at it. It only checks the existence of `openface/nn4.small2.v1.t7`, but not `openface/celeb-classifier.nn4.small2.v1.pkl`. So the md5sum didn't find it.

tesYolan, 2016-07-08:
@wenwei-dev It deletes it if it's lost as in https://github.com/cmusatyalab/openface/blob/master/models/get-models.sh#L50 I guess this error should be an indication to rerun hrtool -i again; Or incoporate something like https://github.com/hansonrobotics/HEAD/blob/656fa117721d9f5e8405d4a2acd26d42521c3191/scripts/hrtool#L241 into the code so that we skip unnecessary steps and only get things that are lost.

wenwei-dev, 2016-07-08:
Once it's deleted, it won't download it again even you run get-model.sh, unless you delete the directory (https://github.com/cmusatyalab/openface/blob/master/models/get-models.sh#L36).

tesYolan, 2016-07-08:
We need to suppress the error from blocking the rest of the installation then as it checks for a file that is the one that we need so it's not essential to make it pass. But that requires the we change their code and create a pull request; But the the easiest thing to do is just delete the folder and do it again or we need to modify their code. @wenwei-dev what do you think to close this issue?

wenwei-dev, 2016-07-08:
I tend to like to fork it and fix the script. Let the error pass silently
is bad. We will have error when we run it and it's hard to find the cause
sometimes. And the advantage of forking is we have a full control of the
code.

Yeah, delete the file and do again when there is error may be the easiest
way.

tesYolan, 2016-07-08:
@wenwei-dev Okay; i forked it in https://github.com/hansonrobotics/openface i would update the conditions and close this issue afterwards/

tesYolan, 2016-07-08:
@wenwei-dev Forked fixed and closed. Though it won't happen now; People need to remove openface that has been already forked from cmu repo. It won't be an issue right?

closed 2016-07-08 by tesYolan

wenwei-dev, 2016-07-10:
Thanks. There is a function in hrtool that do this kind of update. The
function name is "do_migrate". If people run "hrtool -u", it will be
called. So the remote url is updated to the new one. You can also call it
with "hrtool -m".

PR #68 (2016-07-08): Vision tools
Fixes #63. Run ./hrtool -i will fix the undownloaded files.