hansonrobotics/HEAD #120: /tmp/hrtool -B weirdness

linas, 2016-07-27:
When running `/tmp/hrtool -B`  I get this unexpected message:
```
# /tmp/hrtool -B
[WARN] The workspace configured /home/hansonrobotics doesn't match the your working path /
Continue using /home/hansonrobotics? [y/N]
```

I had previously done
```
/tmp/hrtool -w  /home/hansonrobotics
```

so I would expect that the configured workspace really would be that, and not `/`

wenwei-dev, 2016-07-29:
Added to README  https://github.com/hansonrobotics/HEAD#workspace-warning+
If you are using orphan hrtool like /tmp/hrtool, you can ignore it.

linas, 2016-07-29:
why do you say "orphan"? Is it not supported? All of the rest of the README says that you *should* use `/tmp/hrtool`, so I don't understand why its considered to be orphaned.  If this is not the right tool, then where can the correct tool be found?

wenwei-dev, 2016-08-03:
/tmp/hrtool is mainly used to clone the source code. It plays a role like deploying source code. After that you can discard it and use the one HEAD/scripts/hrtool. They are the same.

linas, 2016-08-07:
The readme still says "orphan" without explaining what that means, or how to work around the issue.  Maybe it should simply state that, after the `/tmp/hrtool -G` step, the next step would be to `rm /tmp/hrtool` and them after that, to use `<workspace>/scripts/hrtool -B`  -- this will avoid the error message entirely, and thus will not need a clarifying explanation.

linas, 2016-09-06:
OK, fixed the `README.md` file to say the right thing. Closing.

closed 2016-09-06 by linas