hansonrobotics/HEAD #121: multiple build errors in HEAD/src/vision/ros_nmpt_saliency

linas, 2016-07-27:
Some of the build errors are potentially fatal:
```
HEAD/src/vision/ros_nmpt_saliency/lib/BlockTimer.cpp: In static member function 'static timertype BlockTimer::getCurrentTimeInMicroseconds()':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/BlockTimer.cpp:82:41: warning: integer overflow in expression [-Woverflow]
  timertype microsecondsPerDay = 1440*60*1000000; 
                                       ^
```

Some are merely warnings:
```
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/ImagePatch.cpp: In member function 'void ImagePatch::setImage(const IplImage*, int, int)':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/ImagePatch.cpp:211:49: warning: 'void ImagePatch::setImage(const IplImage*, CvRect, int, int)' is deprecated (declared at /home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/ImagePatch.cpp:205) [-Wdeprecated-declarations]
  this->setImage(image, roi, setData, setIntegral); 
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp: In member function 'virtual std::vector<Feature*> Feature::getSimilarFeatures(int)':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:86:44: warning: 'virtual void Feature::setFeatureParameters(const CvMat*)' is deprecated (declared at /home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:32) [-Wdeprecated-declarations]
   retval[i]->setFeatureParameters(newParams); 
                                            ^
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp: In member function 'void Feature::setRandomParameterValues()':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:118:29: warning: 'virtual void Feature::setFeatureParameters(const CvMat*)' is deprecated (declared at /home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:32) [-Wdeprecated-declarations]
  setFeatureParameters(change); 
                           ^
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp: In member function 'Feature* Feature::copy() const':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:192:53: warning: 'virtual void Feature::setFeatureParameters(const CvMat*)' is deprecated (declared at /home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:32) [-Wdeprecated-declarations]
  patchFeature->setFeatureParameters(this->parameters); 
                                                     ^
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp: In function 'std::istream& operator>>(std::istream&, Feature*&)':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:307:36: warning: 'virtual void Feature::setFeatureParameters(const CvMat*)' is deprecated (declared at /home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/Feature.cpp:32) [-Wdeprecated-declarations]
  model->setFeatureParameters(params); 
                                 ^
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/GentleBoostCascadedClassifier.cpp: In member function 'void GentleBoostCascadedClassifier::setHardNegativeTrainingExamplesFromBGImages()':
/home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/lib/GentleBoostCascadedClassifier.cpp:378:48: warning: 'virtual void GentleBoostCascadedClassifier::setTrainingSet(const std::vector<ImagePatch*>&, const CvMat*)' is deprecated (declared at /home/hansonrobotics/HEAD/src/vision/ros_nmpt_saliency/include/ros_nmpt_saliency/GentleBoostCascadedClassifier.h:263) [-Wdeprecated-declarations]
  setTrainingSet(trainingPatches, trainingLabels); 

```

linas, 2016-07-27:
errors in the face tracker:
```
/home/hansonrobotics/HEAD/src/vision/icog_face_tracker/src/3rdparty/cvblobs/BlobResult.cpp: In member function 'void CBlobResult::PrintBlobs(char*) const':
/home/hansonrobotics/HEAD/src/vision/icog_face_tracker/src/3rdparty/cvblobs/BlobResult.cpp:939:112: warning: format '%f' expects a matching 'double' argument [-Wformat=]
      i, area[i], perimetre[i], externPerimeter[i], perimetreConvex[i], exterior[i], compacitat[i], longitud[i] );
                                                                                                                ^
/home/hansonrobotics/HEAD/src/vision/icog_face_tracker/src/libopentld/tld/TLD.cpp: In member function 'void tld::TLD::writeToFile(const char*)':
/home/hansonrobotics/HEAD/src/vision/icog_face_tracker/src/libopentld/tld/TLD.cpp:441:74: warning: format '%d' expects argument of type 'int', but argument 3 has type 'std::vector<tld::NormalizedPatch>::size_type {aka long unsigned int}' [-Wformat=]
     fprintf(file, "%d #Positive Sample Size\n", nn->truePositives->size());
                                                                          ^
/home/hansonrobotics/HEAD/src/vision/icog_face_tracker/src/libopentld/tld/TLD.cpp:460:75: warning: format '%d' expects argument of type 'int', but argument 3 has type 'std::vector<tld::NormalizedPatch>::size_type {aka long unsigned int}' [-Wformat=]
     fprintf(file, "%d #Negative Sample Size\n", nn->falsePositives->size());
                                                                           ^
/home/hansonrobotics/HEAD/src/vision/icog_face_tracker/src/libopentld/tld/TLD.cpp:509:53: warning: format '%d' expects argument of type 'int', but argument 3 has type 'std::vector<tld::TldExportEntry>::size_type {aka long unsigned int}' [-Wformat=]
         fprintf(file, "%d #numLeaves\n", list.size());
                                                     ^
```

linas, 2016-07-27:
in manyears:
```
/home/hansonrobotics/HEAD/src/sound/manyears_ros/src/manyears_node.cpp: In member function 'bool manyears_ros::ManyEarsNode::parseParams(const ros::NodeHandle&)':
/home/hansonrobotics/HEAD/src/sound/manyears_ros/src/manyears_node.cpp:162:107: warning: format '%i' expects a matching 'int' argument [-Wformat=]
                 ROS_ERROR("Pos element of microphone %i is not a struct, "
                                                                                                           ^
/home/hansonrobotics/HEAD/src/sound/manyears_ros/src/manyears_ros.cpp: In constructor 'manyears_node::many_ears::many_ears(ros::NodeHandle)':
/home/hansonrobotics/HEAD/src/sound/manyears_ros/src/manyears_ros.cpp:165:140: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
                 outputInit(libraryContext_.myOutputSeparated, libraryContext_.myParameters, NULL, NULL, "/tmp/source_sep_****.wav", '*');
                                                                                                                                            ^
/home/hansonrobotics/HEAD/src/sound/manyears_ros/src/manyears_ros.cpp:166:143: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
                 outputInit(libraryContext_.myOutputPostfiltered, libraryContext_.myParameters, NULL, NULL, "/tmp/source_post****.wav", '*');

```

tesYolan, 2016-07-28:
All of the codes that you noticed the warning's are their own library which have been added to the HR repo. I don't think any person in HR have actually modified or touched those particular codes.  And if desired to suppress those particular errors it's better to take those repo out in to their own repository or work from the initial repo.

natnaelargaw, 2016-07-28:
I couldn't find any persuading reason that caused - potentially fatal warning (long long timertype). the variable is long long and it can hold upto 17 digits. any way, addresssed it by replacing the previous multiplication statement by the its own final product and incorporated LOC to get rid of depreciated warnings . checkout nmpt branch.

linas, 2016-07-29:
what does "any person in HR have actually modified or touched those particular codes" mean?  Someone wrote that code -- who?  You? Someone else?  Whoever wrote that code should also make sure that it builds cleanly ...

tesYolan, 2016-07-29:
What i was trying to get at is; 

1. The ManyEar's code code is really (if i am not mistaken) the code that rests; 

https://github.com/introlab/irl_audio/tree/master/manyears_ros
During development and integration they included the code to the HR repo for building the dependent packages that mandeep and naty are working on. 

2. The ros_nmpt code is also really the code 

https://github.com/geni-lab/ros_nmpt_saliency

(which coincidentally mandeep is also the contributor) incoporated in the HR repo during setting up the HEAD repo. 

3. The icog-tracker is also particulary quick way in order to incoporate the opentld as library in the tracker in a quick way by another person that worked on it a while back.  ( The opentld is a previous work of the guys who wrote the CppMT). Also the original library code is located at: 

https://github.com/gnebehay/OpenTLD/

I didn't do any of this aforementioned jobs and was explaining the state how things where done by various other parties.

linas, 2016-07-29:
> What i was trying to get at is;
>
> 1. The ManyEar's code code is really (if i am not mistaken) the code that
> rests;
>
> https://github.com/introlab/irl_audio/tree/master/manyears_ros
> During development [...]

We should NOT be copying other peoples code into HEAD!  This is a time-bomb
waiting to explode.  IED - improvised explosive device, I think they call
them in the middle east. If we need to make changes to that code, we
could/should create a clone, e.g. to
https://github.com/hansonrobotics/irl_audio  Under no circumstances is it
ever correct to copy other people's code into HEAD !!

>
> 2. The ros_nmpt code is also really the code
>
> https://github.com/geni-lab/ros_nmpt_saliency
>
> (which coincidentally mandeep is also the contributor) incoporated in the
> HR repo during setting up the HEAD repo.

Same remarks as above -- this is a basic operating error.

> 3. The icog-tracker is also particulary quick way in order to incoporate
> the opentld as library in the tracker in a quick way by another person that
> worked on it a while back.  ( The opentld is a [...]

Same remarks as 1,2 -- the problem is exactly the "other parties" problem
-- one looses all ability to track the master branches, merge outstanding
pull requests, view the change history, create issues/bug reports in git to
report problems ... everything that makes git and github useful and
interesting and important is completely lost, completely erased, the minute
you copy other peoples repos!

Since I'm already being bombastic, I may as well add  -- the reason that
git was invented was to solve the repository management problems that arise
when working with multiple authors. Copying code like this is working in
exactly the opposite direction of the basic idea of git and of source-code
control in general!

--linas