hansonrobotics/HEAD #130: More permission-denied errors

linas, 2016-07-29:
Per last-nights email:

after a clean install, and creating a a password for user `ubuntu`, the install works but the build breaks with this error message:
```
build-project:
     [echo] relex: /home/ubuntu/hansonrobotics/opencog/relex/build.xml
    [javac] Compiling 1 source file to /home/ubuntu/hansonrobotics/opencog/relex/bin

jar:
      [jar] Building jar: /home/ubuntu/hansonrobotics/opencog/relex/relex-1.6.2.jar

install:
     [copy] Copying 1 file to /usr/local/share/java

BUILD SUCCESSFUL
Total time: 2 seconds
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
CMake Error: Could not open file for write in copy operation /opt/hansonrobotics/vision/CppMT/cmake_uninstall.cmake.tmp
CMake Error: : System Error: Permission denied
CMake Error at CMakeLists.txt:12 (configure_file):
  configure_file Problem configuring file

-- Configuring incomplete, errors occurred!
See also "/opt/hansonrobotics/vision/CppMT/CMakeFiles/CMakeOutput.log".
CMake Error: Cannot open file for write: /opt/hansonrobotics/vision/CppMT/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /opt/hansonrobotics/vision/CppMT/CMakeCache.txt
CMake Error: : System Error: Permission denied
ubuntu@hr-HEAD-2:~$ 
```

The problem is that the directories `/opt/hansonrobotics`, and everything in `/opt` is owned by root, after the `/tmp/hrtool -i` step.

linas, 2016-07-29:
To be clear: in a normal linux system, the /opt directories are always owned by root, not by users, and so CppMT should not be trying to write log files into that location ... this is a bug in the CppMT code.

If the hrtool places directories in /opt owned by the user, instead of root, that's an hrtool bug...

tesYolan, 2016-07-29:
I think the https://github.com/hansonrobotics/HEAD/blob/master/scripts/hrtool#L446 should have handled it. It's not particular a problem of the CppMT. I think for whatever reason that step is being skipped when it's invoked.

linas, 2016-07-29:
I'm re-installing from scratch, to see if it happens again. Its possible that this is due to bad procedure -- I had previously run `/tmp/hrtool -i` as root.

linas, 2016-07-30:
The re-install from scratch seems to avoid this problem.

wenwei-dev, 2016-08-03:
Okay. I close it.

closed 2016-08-03 by wenwei-dev