hansonrobotics/HEAD #132: ImportError: No module named libxml2

linas, 2016-07-31:
I beleive the needed pre-req libxml2 is not being installed by hrtool.  The error message on terminal 9, when running `./dev.sh --oc --cmt` its this:
```
libdc1394 error: Failed to initialize libdc1394
[camera-1] process has died [pid 11293, exit code 1, cmd /opt/ros/indigo/lib/usb_cam/usb_cam_node __name:=camera __log:=/home/ubuntu/.hr/log/a6d76866-56b5-11e6-b8a8-00163ed4b919/camera-1.log].
log file: /home/ubuntu/.hr/log/a6d76866-56b5-11e6-b8a8-00163ed4b919/camera-1*.log
Traceback (most recent call last):
  File "/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/scripts/face_predictor.py", line 17, in <module>
    from image_scraper import image_scaper
  File "/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/scripts/image_scraper.py", line 13, in <module>
    import libxml2
ImportError: No module named libxml2
[camera/face_recognizer-5] process has died [pid 11306, exit code 1, cmd /home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/scripts/face_predictor.py __name:=face_recognizer __log:=/home/ubuntu/.hr/log/a6d76866-56b5-11e6-b8a8-00163ed4b919/camera-face_recognizer-5.log].
log file: /home/ubuntu/.hr/log/a6d76866-56b5-11e6-b8a8-00163ed4b919/camera-face_recognizer-5*.log
Waiting for service /camera/face_recognizer/set_parameters...
libdc1394 error: Failed to initialize libdc1394
```

tesYolan, 2016-07-31:
#133 Should fix it temporaryly by moving the image scraper to non executing section; i would update and close this issue once i include in hrtool the relevant scripts.

linas, 2016-07-31:
Hmm.  now that I plugged in the webcam, the face tracker seems to work despite this eror: I get
```
process[camera/cmt_tracker_view-8]: started with pid [5819]
libdc1394 error: Failed to initialize libdc1394
process[camera/emotime_node-9]: started with pid [5821]
libdc1394 error: Failed to initialize libdc1394
libdc1394 error: Failed to initialize libdc1394
libdc1394 error: Failed to initialize libdc1394
libdc1394 error: Failed to initialize libdc1394
Traceback (most recent call last):
  File "/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/scripts/face_predictor.py", line 17, in <module>
    from image_scraper import image_scaper
  File "/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/scripts/image_scraper.py", line 13, in <module>
    import libxml2
ImportError: No module named libxml2
[camera/face_recognizer-5] process has died [pid 5793, exit code 1, cmd /home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/scripts/face_predictor.py __name:=face_recognizer __log:=/home/ubuntu/.hr/log/117fa5f8-5750-11e6-9baa-00163ed4b919/camera-face_recognizer-5.log].
log file: /home/ubuntu/.hr/log/117fa5f8-5750-11e6-9baa-00163ed4b919/camera-face_recognizer-5*.log
Waiting for service /camera/face_recognizer/set_parameters...
Enter New Face Started
Enter New Face Finished
```

wenwei-dev, 2017-09-07:
Obsolete. Closing.

closed 2017-09-07 by wenwei-dev