hansonrobotics/HEAD #167: system is confused about torch

demaris, 2016-08-19:
I installed by wget /tmp/hrtool, 
then hrtool -i
then hrtool -GB.
I can see lines in console during the build saying [WARN] Skipping Torch installation.

When I run .dev/sh
/opt/hansonrobotics/vision/torch/install/bin/torch-activate: no such file or directory

demaris, 2016-08-19:
/opt/hansonrobotic/vision/torch/ exists, but not install/bin
I ran ./install.sh in that directory, and after a build process the install/bin exists and dev.sh env.sh works.

tesYolan, 2016-08-22:
Did it finish the installation without interruption. As that step is included in the https://github.com/hansonrobotics/HEAD/blob/master/scripts/hrtool#L272 . One possible reason is that it cloned the repo but interrupted the isntallation process it has a chance to finish the installation process. I would look in to making sure that torch-activate is installed before skipping the installation.

wenwei-dev, 2016-11-20:
Looks like the installation was not finished. And it works after running ./install.sh, so closing.

closed 2016-11-20 by wenwei-dev