hansonrobotics/HEAD #203: Fatal ERROR in cmt

linas, 2016-09-06:
The build of CMT fails, and the system is not usable as a result. Ther is a warning just before the fatal error:
```
Warnings   << cmt_tracker:cmake /home/ubuntu/hansonrobotics/HEAD/logs/cmt_tracker/build.cmake.000.log
CUDA_TOOLKIT_ROOT_DIR not found or specified
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:409 (message):
  catkin_package() include dir
  '/home/ubuntu/hansonrobotics/HEAD/build/cmt_tracker' should be placed in
  the devel space instead of the build space
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  CMakeLists.txt:111 (catkin_package)
```

After this comes the error:
```
Errors     << cmt_tracker:make /home/ubuntu/hansonrobotics/HEAD/logs/cmt_tracker/build.make.000.log
In file included from /home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/src/face_locator_node.cpp:1:0:
/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/include/face_locator_node.h:46:23: fatal error: Flandmark.h: No such file or directory
 #include "Flandmark.h"
                       ^
compilation terminated.
In file included from /home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/src/cmt_tracker_node.cpp:2:0:
/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/include/face_locator_node.h:46:23: fatal error: Flandmark.h: No such file or directory
 #include "Flandmark.h"
                       ^
compilation terminated.
make[2]: *** [CMakeFiles/face_locator_node.dir/src/face_locator_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/face_locator_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/cmt_tracker_node.dir/src/cmt_tracker_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/cmt_tracker_node.dir/all] Error 2
make: *** [all] Error 2
```

wenwei-dev, 2016-09-06:
One of the dependencies is not installed or successfully installed. You can try to reinstall vision tool dependencies using 'hrtool -r install_vision_deps', or simply rerun 'hrtool -i' to fix other potential missing file error.

linas, 2016-09-07:
A lot of `hrtool` is piped to /dev/null, so its basically impossible to debug.  Running hrtool just sometimes seems to hang for minutes at a time -- its doing something, but you can't tell what, and you can't tell if it got an error or not.  Can this be removed?

Basically, hrtool hits errors a lot, and I'm spending a lot of time trying to figure out why.

linas, 2016-09-08:
Closing.  Repeated cycles of `hrtool -i` and `hrtool -B` seems to have fixed this issue.

closed 2016-09-08 by linas