hansonrobotics/HEAD #204: Wild gesticulation in blender

linas, 2016-09-06:
This has been an issue for me, in blender, for many many months, and has not gone away:

When running blender, her head wildly shakes from side to side, turning her nec by more than 45 degrees, 2-3 times per second, constantly, without stopping. When she falls asleep, she continues to do this, but with her eyes closed.   Ther does not seem to be any button in the control GUI to stop this:  I tried turning everything off, she still does this.

For me, for development, this is worse than useless -- it is impossible to see what she does -- it drowns out most facial expressions, which are overwhelmed by the side-to-side shaking, and it make it impossible to test nodding and shaking.

linas, 2016-09-07:
Interesting -- it looks good, when running blender version 2.76  -- however, hrtool installs version 2.71 which looks really bad.

linas, 2016-09-07:
Using `ppa:thomas-schiex/blender` instead of `ppa:irie/blender` in hrtool installs a newer version of blender.  However, the newer version has errors:
```
blender_api/rigControl/commands.py: line 154
SetGesture ... 
AttributeError: module 'bpy' has no attribute 'evaAnimationManager'.
```

linas, 2016-09-07:
More errors, that might be causing this:
```
Traceback (most recent call last):
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/Sophia.blend/loader.py", l
ine 15, in <module>
    register()
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/Sophia.blend/loader.py", l
ine 13, in register
    rigControl.init()
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/rigControl/__init__.py", l
ine 47, in init
    animationManager.init()
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/rigControl/animationManage
r.py", line 602, in init
    bpy.evaAnimationManager = AnimationManager()
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/rigControl/animationManage
r.py", line 77, in __init__
    self.actuatorManager = ActuatorManager()                            [1459/1493]
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/rigControl/actuators.py", 
line 83, in __init__
    import artistic.actuators
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/artistic/actuators.py", li
ne 2, in <module>
    import numpy as np
  File "/usr/lib/python3/dist-packages/numpy/__init__.py", line 153, in <module>
    from . import add_newdocs
  File "/usr/lib/python3/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/lib/python3/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/lib/python3/dist-packages/numpy/lib/type_check.py", line 11, in <modul
e>
    import numpy.core.numeric as _nx
  File "/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 6, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'
Command Source 'ros' won't build
No Command Sources found

```

linas, 2016-09-07:
According to goolge and various mailing lists, numpy has not yet been ported to python 3.whatever.  We probably should not be using numpy with blender, if we want this stuff to work.

linas, 2016-09-07:
Stubbing out the entire contents of "artistic", so that the numpy error is avoided, does get blender to power up, and gets her to do keepalive motions. However, animations still won't work, I still get:
```
Dependency cycle detected:
  deform depends on control through Child Of.
  control depends on deform through Locked Track.

Command Source 'ros' won't build
No Command Sources found
Traceback (most recent call last):
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/rigControl/blenderPlayback.py", line 85, in execute
    logger.info(eval(self.action))
  File "<string>", line 1, in <module>
  File "/home/ubuntu/hansonrobotics/HEAD/src/blender_api/rigControl/commands.py", line 14, in init
    bpy.ops.wm.animation_playback()
  File "/usr/share/blender/2.78/scripts/modules/bpy/ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Operator bpy.ops.wm.animation_playback.poll() failed, context is incorrect

location: <unknown location>:-1

location: <unknown location>:-1
```

demaris, 2016-09-07:
I was about to try to use 1.76 in order to get radeon gpu acceleration.  I
seem to have some memory that you have to start blender and put some
startup script in the text editor or some fiddly thing like that.
I once got a newer version to work, will check back through emails and
slack for any insight.

linas, 2016-09-07:
Ahh .. OK .. more debugging results:  the error message 
```
Command Source 'ros' won't build
No Command Sources found
```

really does mean that ROS is not attached to blender. Thus, when she comes up, she will just do breathing motions, just fine (after killing the numpy bug) . She will also smile, frown, etc.  ***But if you make her look left*** (or right, or up or down...)  then her head motions will go completely wild. This happens on blender 2.78, and looks just like the bug in blender 2.71

So, this issue is unrelated to the blender version --  its instead due to some transform bug in the blender file itself.

I don't understand how anyone can get any work done. This is a show-stopper, dead-in-your-tracks issue. For me, at least, the whole rig is unusable.  (My work-around is to use the old blender model, from before summer 2015, back when things still worked.  That old model still does work... ignoring that some animation names were changed)

demaris, 2016-09-08:
I'm not sure how you're controlling the head, but I think I see what you're
talking about, and I can get it from the blender menu controls but also
from the webui puppeteering head control.  I think the latter has to be
going through ros and the ros command layer.
What it seems to me is different is that those commands used to move the
head and keep it there, but now something is resetting the head rapidly
back to some previous or new position roughly centered as soon as there's
no active event.  If you stop a head direction gesture, it snaps back; if
you try to move the widget around it jerks to center and back to the
position.  There is some way more animated keepalive than was there back in
say November 2015; maybe that's stealing the head direction.  I don't know
who took over that animation when I shifted to puppeting and chat.

vytasrgl, 2016-10-25:
The only thing i can think of is that one very slow machines if FPS < 5 it might be that more animations are queued up than blender can actually play so multiple commands being executed at same time.
Otherwise it might be that 
Are there any external nodes that publish to /blender_api/setFaceTarget and what distance (x coordinate it publishes).
@demaris if behavior tree running and head is being moved left or right manual the behavior tree will reset its position as keep alive part of the tree will move head randomly. To manual control the head you may need turn off behavior tree.

linas, 2016-11-10:
I have a brand-new graphics card now, will review.

sameerthahir, 2018-02-22:
Hello, I am stuck with the same error.

Command Source 'ros' won't build
No Command Sources found.

Is there any update on a fix