hansonrobotics/HEAD #205: Xenial -- cmake is not installed

linas, 2016-09-06:
While testing on xenial, I noticed that `cmake` is not installed, so that `hrtool -B` fails.

linas, 2016-09-06:
Never mind; fixed in @496fe839f456935c1ddd56453c65177d99e5f360

linas, 2016-09-06:
Re-opening -- there are more install issues -- binutils is not installed, bfd is not installed, boost is not installed.

linas, 2016-09-06:
bfd.h is provided by binutils-dev
boost requires libboost-dev
libboost-date-time-dev libboost-filesystem-dev
libboost-program-options-dev
libboost-regex-dev libboost-serialization-dev libboost-system-dev
libboost-thread-dev

linas, 2016-09-06:
Sigh. Also, python and guile are not installed. I don't understand how there can be this many breakages ...

wenwei-dev, 2016-09-06:
All the software we are using is aiming for Ubunu 14.04. If you want to run it on Xenial, for sure it won't work because nobody has tested it. At least blender won't work. It's a PPA version.

wenwei-dev, 2016-11-20:
Closing because Xenial is not supported.

closed 2016-11-20 by wenwei-dev