hansonrobotics/HEAD #217: apt-get failure in lxc

StrangeTcy, 2016-09-08:
I've tried to create an lxc container on my Ubuntu 16.04. It got created, and I started it and attached to it and tried to follow the instructions from README and to `apt-get install openssh-server`, but that resulting in a lot of `failed to fetch` errors.
Trying obvious things showed that such commands as `wget`, `curl` or `telnet` were not found. The best I could manage was a `ping`.

Did I do anything wrong? Is there an issue with lxc on Ubuntu?

linas, 2016-09-09:
lxc works great in ubuntu. I would go through the basic lxc tutorials and make sure you've got the network configured for the thing.  No networking means no apt-get.

Also: please note: HEAD only works for ubuntu 14.04 so you want to load trusty 14.04 in the lxc container, not xenial 16.04.

linas, 2016-09-09:
closing, because this is not really a HEAD issue -- its some basic LXC setup issue. Perhaps the README should mention it, though.

closed 2016-09-09 by linas

StrangeTcy, 2016-09-10:
It turned out to be an issue with my host system (Ubuntu Xenial), not the system on the container being created (Ubuntu Trusty).

As for the `README`, in the `LXC Install, build and run directions.` section it explictily states 
`lxc-create -t download -n my-hr-HEAD -- -d ubuntu -r trusty -a amd64`
which should dispel any doubts.