hansonrobotics/HEAD #224: Editing files

StrangeTcy, 2016-09-10:
`README`, in the `LXC Install, build and run directions.` section, suggests installing `openssh-server` and then running 
`vi authorized_keys`
I never really learned to use `vi` and prefer text editors with a gui, like gedit.
I can install it from inside the container, but it'd seem I can't run i; it outputs errors when I try:

> root@my-hr-HEAD:/home/ubuntu/.ssh# gedit authorized_keys 
> error: XDG_RUNTIME_DIR not set in the environment.
> 
> (gedit:15637): Gtk-WARNING **: cannot open display: 
> 

I think there're some pretty simple options to running `lxc` that can be used to make the system in the container aware of the available displays, but I'll probably spend more time researching it myself than if they were posted here.

linas, 2016-09-10:
You can't get an x-windows display to the x-server until you install ssh. To install ssh, you need to add your key to the authorized keys file.  You can always just use redirection and work blindly: for example: `cat mykey.pub >> ~/.ssh/authorized_keys`

As a general rule, it would be very useful for you to learn how to use Linux -- and the Linux command line, in general. Having to use a GUI editor is like being forced to sit in a wheel-chair -- very comfortable, for all the places that a wheel chair can go, but it does put very large parts of the computing world off-limits to you.  It's worth taking the effort to learn how to walk -- not easy, not immediately rewarding -- but after some years, you'll be very surprised.

StrangeTcy, 2016-09-13:
Agreed, generally.
But I admit to being scared off vi; I've heard about it by numerous people, and their accounts pretty much amounted to [this](http://www.thejach.com/imgs/vim_learning.jpg).

But I will try to learn it, now that I have such a need for it.

linas, 2016-09-13:
Heh. Well, according to that diagram, vi is the easiest of all to learn: after zero time, you can do everything.  Recall how a learning curve is graphed: amount of elapsed time runs along the x axis, and proficiency is the y axis.  So if you get very good, very quickly, then you get a curve like that for vi.    Compare that to the curve to visual studio -- after using it long enough, you become completely incompetent! 

Perhaps you shouldn't trust whomever drew that chart-- they don't seem to actually know what a learning curve is.

wenwei-dev, 2016-11-20:
It's HEAD irrelevant topic. Closing.

closed 2016-11-20 by wenwei-dev