hansonrobotics/HEAD #277: Webui motor settings page can't open

vytasrgl, 2016-10-06:
http://localhost:8000/#/admin/motors

robot: sophia5
1.js:28setting expression to: Neutral, intensity: 0
1.js:28pointing head:
1.js:28Object {yaw: 0, pitch: 0, roll: 0}
1.js:221 Uncaught TypeError: t.get(...) is not a functiona.Row.e.View.extend.makeCell @ 1.js:221a.Row.e.View.extend.initialize @ 1.js:221e.View @ 1.js:222i @ 1.js:222a.Body.e.View.extend.insertRow @ 1.js:221b @ 1.js:26v @ 1.js:26d @ 1.js:26u.trigger @ 1.js:26n.extend._onModelEvent @ 1.js:26b @ 1.js:26v @ 1.js:26d @ 1.js:26u.trigger @ 1.js:26n.extend.set @ 1.js:26n.extend.add @ 1.js:26(anonymous function) @ 1.js:207s.ajax.success @ 1.js:206l @ 1.js:24c.fireWith @ 1.js:24i @ 1.js:25(anonymous function) @ 1.js:26

wenwei-dev, 2016-10-06:
@grebsle Since Vytas is on vacation, would you like to take a look? This is kind of urgent. I can't open motors to set the parameters. I will need it to change the mapping.

wenwei-dev, 2016-10-08:
Can someone fix this bug? Sorry, it's urgent.

grebsle, 2016-10-10:
https://github.com/hansonrobotics/HEAD/pull/289

closed 2016-10-25 by vytasrgl