hansonrobotics/HEAD #324: fatal error: Flandmark.h: No such file or directory

linas, 2016-11-02:
After doing an `hrtool -U` and `hrtool -B`  I get the following error:
```
_______________________________________________________________________________
Errors     << cmt_tracker:make /home/ubuntu/hansonrobotics/HEAD/logs/cmt_tracker/build.make.000.log
In file included from /home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/src/face_locator_node.cpp:1:0:
/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/include/face_locator_node.h:46:23: fatal error: Flandmark.h: No such file or directory
 #include "Flandmark.h"
                       ^
compilation terminated.

In file included from /home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/src/cmt_tracker_node.cpp:2:0:
/home/ubuntu/hansonrobotics/HEAD/src/vision/cmt_tracker/include/face_locator_node.h:46:23: fatal error: Flandmark.h: No such file or directory
 #include "Flandmark.h"
                       ^
compilation terminated.
```

wenwei-dev, 2016-11-03:
This is a dependency for cmt tracker. Please run `./hrtool -i` to install that and try again.

linas, 2016-11-08:
OK, trying that.

I notice that hrtool says this: `./octool -rdpv >/dev/null`  and `./octool -l >/dev/null` -- can these be changed to not pipe to dev null? It makes it hard to understand what the script is doing at that particular moment.

linas, 2016-11-08:
OK, that seemed to fix the bug

closed 2016-11-08 by linas