hansonrobotics/HEAD #411: Active performance piece, and controls in event sequence

vytasrgl, 2016-12-08:
In readonly(playback mode):
The button of the piece currently should be highlighted.
In editor mode:
1. The sequence line should be highlighted.
2. (TBD) either of:
 a. Clicking on event sequence performance will move the performance marker to that specific performance.
 b. performance pieces  in event sequence has play button to start the queue from that specific point.
 c. a and b.
3. Duration label should be fixed.

For these the service returning nodes from, requested performance IDS needs to also return the time ranges for those performance ids loaded.

grebsle, 2016-12-19:
https://github.com/hansonrobotics/HEAD/pull/463/

vytasrgl, 2016-12-21:
Duration seems to be still not fetched properly, and it makes the clicking on event sequence to go to wrong place on timeline.
Also clicking on first performance starts it, which is inconsistent

vytasrgl, 2016-12-21:
Actually, the issue with autopause that prevent starting the performance immediatly.

vytasrgl, 2016-12-21:
Keep in mind that Pause node is actually 0.1 seconds in timeline.

vytasrgl, 2016-12-29:
Duration label still 0 after performances added.
Please implement point 2C so the clicking on the performance in sequence only move marker, and if button is pressed than it would actually play performance.

grebsle, 2016-12-30:
Duration seems to be calculated properly, need more info to reproduce
https://github.com/hansonrobotics/HEAD/pull/492

vytasrgl, 2017-01-02:
Reproduce:
1. "Clear All" perfromances
2. Refresh page with some performances in the current folder.
3. Press "Add All"

vytasrgl, 2017-01-03:
In addition:
After moving the marker to specific place by clicking on event sequence the nodes with (0.00) start time wont be played after pressing resume button

grebsle, 2017-01-09:
https://github.com/hansonrobotics/HEAD/pull/509

closed 2017-01-10 by vytasrgl