hansonrobotics/HEAD #415: Do not allow save current timeline

vytasrgl, 2016-12-08:
1. While not in edit mode hide the Name field with save button.
2. "Done" should ask if need to be changed (if any made) before closing the edit view (Yes, No, Cancel).
3. In "Edit" mode clicking on other performance should immediately start editing the new performance if no changes are made on current performance.

grebsle, 2016-12-08:
https://github.com/hansonrobotics/HEAD/pull/416

vytasrgl, 2016-12-09:
1. Empty name field and save button  still visible (should be hidden). This was done to allow combine multiple performances however it seems its rarely used so can remove for now.
2. If changes are made on currently edited performance, and user tries to click on other performance or other tab or leave page popup should notify.
3. Edit mode should be triggered only by pressing edit button on sequence. Will need to update.

grebsle, 2016-12-09:
1. The name input and save button already removed in puppeteering. Should it be hidden in performances page too?

vytasrgl, 2016-12-09:
Yes,  it should be hidden once no performance is being edited

grebsle, 2016-12-12:
https://github.com/hansonrobotics/HEAD/pull/428

closed 2016-12-15 by vytasrgl

demaris, 2016-12-15:
Thanks Vytas - as you noticed those things were a bit confusing for me.