hansonrobotics/HEAD #433: Naming new performance rules

vytasrgl, 2016-12-15:
**Empty(new) folder:**
{folder_name}_1
**Existing Folder**
{last_performance_name_part}_{last_performance_number_part + 1}
if last_performance_number_partnot exist:  {last_performance_name_part}_1
if last_performance_name_part: {last_performance_number_part + 1}

grebsle, 2016-12-21:
https://github.com/hansonrobotics/HEAD/pull/469

vytasrgl, 2016-12-22:
last_performance name should be the name of performance that is last in the folder after natural sort. 
Currently in some case if ther are  PG01 ...PG16 performances the new performance name is PG01 01 should be PG17

Also the following should be fixed:
1,2,3,10,A10 should result in "A11" name as its last performance after sort

vytasrgl, 2016-12-23:
after #473 the only issue i found for example: "Slide5" gets you next name as "Slide06" which when is not the last on the list.

vytasrgl, 2016-12-23:
Also since we do natural sort "01" is not necessary, better use 1 instead

grebsle, 2016-12-28:
https://github.com/hansonrobotics/HEAD/pull/481

closed 2017-01-02 by vytasrgl