hansonrobotics/HEAD #510: Node Setting ranges

vytasrgl, 2017-01-10:
Multiple node setting ranges needs to be changed to allow better authoring of performances:
- Animation
  - speed (0.2 -2)
  - magnitude (0.2 -3). For magnitudes that are more than 1, multiple commands need to be sent with magnitudes that adds up to total: for example if magnitude is 2.5 two commands need to be sent with 1 magnitude and the additional with 0.5 magnitude.
- Speech
  - Speed (0.7 - 0.3)
  - Pitch (0.8 -1.2)
- LookAt
  - speed (0.1 - 2)

grebsle, 2017-01-11:
https://github.com/hansonrobotics/HEAD/pull/521

grebsle, 2017-01-11:
Wouldn't it be better to parse animation messages in their node and send multiple commands there in case magnitude is over 1? Because now it would have to be updated both in webui and performance runner

closed 2017-02-16 by vytasrgl