hansonrobotics/HEAD #576: Build problems with hrtool

Narifix, 2017-02-24:
Hello,

I'm having issues with building the Opencog repository correctly when using `./hrtool -B`. The problem appears to be with atomspace but I can't seem to fix it. During the compilation, I get these errors that pass by:

> /home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc: In member function ‘std::vector<opencog::AndBIT>::iterator opencog::BIT::erase(std::vector<opencog::AndBIT>::const_iterator)’:
/home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc:638:26: error: no matching function for call to ‘std::vector<opencog::AndBIT>::erase(std::vector<opencog::AndBIT>::const_iterator&)’
  return andbits.erase(pos);
                          ^
/home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc:638:26: note: candidates are:
In file included from /usr/include/c++/4.8/vector:69:0,
                 from /usr/include/c++/4.8/bits/random.h:34,
                 from /usr/include/c++/4.8/random:50,
                 from /usr/include/c++/4.8/bits/stl_algo.h:65,
                 from /usr/include/c++/4.8/algorithm:62,
                 from /usr/include/boost/iterator/iterator_concepts.hpp:29,
                 from /usr/include/boost/range/concepts.hpp:20,
                 from /usr/include/boost/range/algorithm/binary_search.hpp:15,
                 from /home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc:23:
/usr/include/c++/4.8/bits/vector.tcc:134:5: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::iterator) [with _Tp = opencog::AndBIT; _Alloc = std::allocator<opencog::AndBIT>; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<opencog::AndBIT*, std::vector<opencog::AndBIT> >; typename std::_Vector_base<_Tp, _Alloc>::pointer = opencog::AndBIT*]
     vector<_Tp, _Alloc>::
     ^
/usr/include/c++/4.8/bits/vector.tcc:134:5: note:   no known conversion for argument 1 from ‘std::vector<opencog::AndBIT>::const_iterator {aka __gnu_cxx::__normal_iterator<const opencog::AndBIT*, std::vector<opencog::AndBIT> >}’ to ‘std::vector<opencog::AndBIT>::iterator {aka __gnu_cxx::__normal_iterator<opencog::AndBIT*, std::vector<opencog::AndBIT> >}’
/usr/include/c++/4.8/bits/vector.tcc:146:5: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::iterator, std::vector<_Tp, _Alloc>::iterator) [with _Tp = opencog::AndBIT; _Alloc = std::allocator<opencog::AndBIT>; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<opencog::AndBIT*, std::vector<opencog::AndBIT> >; typename std::_Vector_base<_Tp, _Alloc>::pointer = opencog::AndBIT*]
     vector<_Tp, _Alloc>::
     ^
/usr/include/c++/4.8/bits/vector.tcc:146:5: note:   candidate expects 2 arguments, 1 provided
/home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc: In member function ‘std::vector<opencog::AndBIT>::iterator opencog::BIT::erase(std::vector<opencog::AndBIT>::const_iterator, std::vector<opencog::AndBIT>::const_iterator)’:
/home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc:648:31: error: no matching function for call to ‘std::vector<opencog::AndBIT>::erase(std::vector<opencog::AndBIT>::const_iterator&, std::vector<opencog::AndBIT>::const_iterator&)’
  return andbits.erase(from, to);
                               ^
/home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc:648:31: note: candidates are:
In file included from /usr/include/c++/4.8/vector:69:0,
                 from /usr/include/c++/4.8/bits/random.h:34,
                 from /usr/include/c++/4.8/random:50,
                 from /usr/include/c++/4.8/bits/stl_algo.h:65,
                 from /usr/include/c++/4.8/algorithm:62,
                 from /usr/include/boost/iterator/iterator_concepts.hpp:29,
                 from /usr/include/boost/range/concepts.hpp:20,
                 from /usr/include/boost/range/algorithm/binary_search.hpp:15,
                 from /home/ubuntu/hansonrobotics/opencog/atomspace/opencog/rule-engine/backwardchainer/BIT.cc:23:
/usr/include/c++/4.8/bits/vector.tcc:134:5: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::iterator) [with _Tp = opencog::AndBIT; _Alloc = std::allocator<opencog::AndBIT>; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<opencog::AndBIT*, std::vector<opencog::AndBIT> >; typename std::_Vector_base<_Tp, _Alloc>::pointer = opencog::AndBIT*]
     vector<_Tp, _Alloc>::
     ^
/usr/include/c++/4.8/bits/vector.tcc:134:5: note:   candidate expects 1 argument, 2 provided
/usr/include/c++/4.8/bits/vector.tcc:146:5: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::iterator, std::vector<_Tp, _Alloc>::iterator) [with _Tp = opencog::AndBIT; _Alloc = std::allocator<opencog::AndBIT>; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<opencog::AndBIT*, std::vector<opencog::AndBIT> >; typename std::_Vector_base<_Tp, _Alloc>::pointer = opencog::AndBIT*]
     vector<_Tp, _Alloc>::
     ^
/usr/include/c++/4.8/bits/vector.tcc:146:5: note:   no known conversion for argument 1 from ‘std::vector<opencog::AndBIT>::const_iterator {aka __gnu_cxx::__normal_iterator<const opencog::AndBIT*, std::vector<opencog::AndBIT> >}’ to ‘std::vector<opencog::AndBIT>::iterator {aka __gnu_cxx::__normal_iterator<opencog::AndBIT*, std::vector<opencog::AndBIT> >}’
make[2]: *** [opencog/rule-engine/CMakeFiles/ruleengine.dir/backwardchainer/BIT.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[7 of 9] Compiling OpenCog.AtomSpace.Utils ( OpenCog/AtomSpace/Utils.hs, .stack-work/dist/x86_64-linux/Cabal-1.24.0.0/build/OpenCog/AtomSpace/Utils.o )
[8 of 9] Compiling OpenCog.AtomSpace.Query ( OpenCog/AtomSpace/Query.hs, .stack-work/dist/x86_64-linux/Cabal-1.24.0.0/build/OpenCog/AtomSpace/Query.o )
[9 of 9] Compiling OpenCog.AtomSpace ( OpenCog/AtomSpace.hs, .stack-work/dist/x86_64-linux/Cabal-1.24.0.0/build/OpenCog/AtomSpace.o )
opencog-atomspace-0.1.0.7: copy/register
Installing library in
/home/ubuntu/hansonrobotics/opencog/atomspace/opencog/haskell/.stack-work/install/x86_64-linux/lts-7.3/8.0.1/lib/x86_64-linux-ghc-8.0.1/opencog-atomspace-0.1.0.7-KYIyFxn1Y4uLIf7WFu9IoH
Registering opencog-atomspace-0.1.0.7...
[ 95%] Built target haskell-atomspace-lib
make[1]: *** [opencog/rule-engine/CMakeFiles/ruleengine.dir/all] Error 2
make: *** [all] Error 2
-- Build type: Release
-- Found CogUtil version 2.0.2 at /usr/local/lib/libcogutil.so
-- CogUtil found.
-- AtomSpace not found
CMake Error at CMakeLists.txt:154 (MESSAGE):
  AtomSpace missing: it is needed for everything!
-- Configuring incomplete, errors occurred!

Building atomspace manually results in the same errors. I'm thinking there may have been a problem during the dependency installation (I've tried `hrtool -i` many times)

(I ran `hrtool -B` after doing the steps in the README. I tried using LXC too but got the same error)

I tested dev.sh, and some things work and some don't. The Roscore terminal says Pocketphinx is not found, the Cog terminal says there is no code for module Opencog, and RelEx is not found. What does seem to work, though, are the Blender and vision modules.

I'm currently running this on a Macbook Pro 2011 (8,2) with Ubuntu 14.04.5 installed. I originally tried this on my desktop machine and got the same errors. Could I just be missing a step?

I'm very new to all of this, by the way. I'm sorry if anything I said was somehow irrelevant.

I'd really appreciate the help so I can experiment with this. Thanks!

wenwei-dev, 2017-02-24:
From your description, it does seem some dependencies are missing. Could you please attach the output of terminal when you run `./hrtool -i`.

Narifix, 2017-02-24:
Here's my output file
[output.txt](https://github.com/hansonrobotics/HEAD/files/799083/output.txt)

Narifix, 2017-02-24:
Oops, my mistake. I think the output file I just posted didn't include everything from the terminal. I'm attaching the real one here.
[output.txt](https://github.com/hansonrobotics/HEAD/files/799187/output.txt)

Narifix, 2017-02-24:
Here is the output from the terminal when running `./hrtool -i` (attached):
[output1.txt](https://github.com/hansonrobotics/HEAD/files/799226/output1.txt)

wenwei-dev, 2017-03-01:
From what I see, all the dependencies are installed. There is an error about missing prim_type.h, but that shouldn't stop atomspace form compiling. The missing file should be in "/opt/hansonrobotics/include/sphinxbase/". Maybe because of your environment settings, the include path is not added to c compiler. That could explain why atomspace is not built.

linas, 2017-03-01:
Sorry for late reply, just saw this. Nil Geisweiller has been messing with
BIT and seems to have broken something. Maybe its fixed already.

Linas.

linas, 2017-03-01:
Please update, I'm pretty sure it's fixed.

Nil

closed 2017-04-18 by wenwei-dev