hansonrobotics/HEAD #618: hrtool not building opencog/ros-behavior-scripting

eddiemonroe, 2017-03-30:
Perhaps this issue should be in the ros-behavior-scripting repo, not sure.

With a fresh install of HEAD stack (and "./hrtool -d"), ./dev.sh --oc gives an error when running btree-psi.scm in the cog tmux terminal: 
"ERROR: no code for module (opencog movement)"

I believe this is due to ros-behavior-scripting not building correctly

./hrtool -B leads to error when attempting to build ros-behavior-scripting:

CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/eddie/hansonrobotics/opencog/ros-behavior-scripting/build/CMakeFiles/CMakeOutput.log".

From looking at the ros-behav-scripting README, perhaps hrtool build script needs to be adjusted for build of ros-behavior-scripting using catkin?

leungmanhin, 2017-03-31:
Maybe try doing `source /opt/ros/indigo/setup.bash` before building?

wenwei-dev, 2017-03-31:
From the error, catkin is not found. So I suggest you run ./hrtool -i to
make sure all the dependencies are installed.

linas, 2017-03-31:
I also got the "catkin not found" error, and running setup.bash fixed it. 

I think that the core issue is that, from now on out, the ros-behavior-scripting code should be built in the same ROS/catkin home/build directory as all the other hanson-robotics ROS nodes.  All the opencog code was removed, and all that's left are some ROS nodes and some messages.

eddiemonroe, 2017-03-31:
Running setup.bash fixed the build error for me as well.

But, I still get 
  "ERROR: no code for module (opencog movement)"
when running dev.sh --oc

Looks like movement.scm (and webui.scm) are being installed in /scm/opecog/. 
Should these be installed in /usr/local/share/opencog/scm/opencog/ instead?
Or alternatively /scm/opencog/ needs to be added to the scm load path?

linas, 2017-03-31:
that install path is wrong, it needs to be in usr/local  maybe the cmakefile has a typo in it?

linas, 2017-04-01:
I just tried fixing this now, but I couldn't -- I'm hoping that @wenwei-dev Wenwei can find a good solution.   Let me describe the problem, first, though:

The original issue was bug opencog/ros-behavior-scripting#162 -- that ros-behavior-scripting depended on ROS message types defined only in HEAD.   Until HEAD is modularized, the only solution  I could think of was to copy the message types into ros-behavior-scripting.  Which I did in pull req   opencog/ros-behavior-scripting#167 However, that seems to have broken the opencog install -- ROS tries to take over the install process, and instead of putting files into `/usr/local` it puts them somewhere else.

I don't quite know how to fix this.   A partial work-around would be to undo some of the changes in 167, but that then causes different problems. I'm stumped.

linas, 2017-04-01:
One problem is that if one says `sudo make install`, then ROS tries to write crazy stuff into `/usr/local/` thus breaking the standard unix FHS.  ROS uses it's own, unique FHS that isn't compatible with unix.

wenwei-dev, 2017-04-03:
Yes, as you said, ROS has its own FHS. Unless you use standard cmake file,
there is no workaround. But then you have to find a complex way to compile
the ros messages.

So we should maybe put all the message types together as a single ROS
package called say hr_msgs, just like std_msgs. In this way, you just write
"from hr_msgs.msg import ChatMessage". Then the whole
ros-behavior-scripting package is a standard python package + scm package.

linas, 2017-04-03:
But Wenwei, if you look at the cmake file, it is "standard" -- its a cut-n-paste of what the other ROS cmakefiles look like.   ... mostly.  But it has some bug, and I cannot spot the bug.

wenwei-dev, 2017-04-05:
It's not "standard" cmake fille, because it uses some custom catkin functions and macros, such as "add_message_files". But it works if you pass CMAKE_INSTALL_PREFIX to cmake command. For example `cmake -DCMAKE_INSTALL_PREFIX=/usr/local/opencog -DDATADIR=/usr/local/opencog/data ..`. 

I also think we should work out a common place to install ROS packages in the hanson-opencog environment, so that both OpenCog and HEAD can find.  Eventually we need to install all the ROS packages to such place, rather than build in "devel" in each repository.

linas, 2017-04-05:
So:

1) I just copied the cmakefile that is used in HEAD. So its the same thing here, as there.  If its broken here, then its broken there, too.

2) Asking for `-DCMAKE_INSTALL_PREFIX=` is patently absurd/broken.  The default install location is `/usr/local/` and there is only one reason whatsoever to over-ride this: for a distro, e.g. redhat, centos, ubuntu to change the prefix to `/usr/`.  Any other use of prefixes is pretty much a gross mis-use of them. 

3) We really, really need to split up HEAD into multiple components. It is completely unusable, in its current state.  What I was trying to here was to define message files independently of HEAD, precisely because HEAD is broken. We have to kill HEAD, its the only way.  I cannot repeat this often enough or loudly enough, it seems. HEAD is fundamentally broken.

wenwei-dev, 2017-04-05:
HEAD doesn't install ROS packages to `/usr/local`, and it doesn't need to. It builds packages in `devel`, and install them to `install` next to `devel`. That's why every time you need to `source xxx/devel/setup.bash`, this will setup the correct PATH and other env variables to include `devel`. 

Before we don't build ros-behavior-scripting. We just add the path to PYTHONPATH the modules can be found by python. But now ros-behavior-scripting has become a ROS package, `/usr/local` is not a preferred path for ROS package. I think this is the main problem. To let python discover this location, you have to source the setup.bash file. 

In order to reduce the need to run too many `source` command, we should find a common place, and install all ROS packages to that place.

wenwei-dev, 2017-04-05:
Specifying `-DCMAKE_INSTALL_PREFIX` can be avoid if you put it to CMakeLists.txt.

wenwei-dev, 2017-04-10:
@linas I've created a ROS message package that contains all the current ROS message classes. It's a standalone package called hr_msgs (https://github.com/hansonrobotics/hr_msgs). It will be shared across other packages. 

I've crafted a primitive management tool, which is going obsolete the current hrtool (I hope). You can use it to install hr_msgs by `hr install hr_msgs`. You can safely uninstall it by `hr uninstall hr_msgs`. 

The default path of hr_msgs is `/opt/hansonrobotics/ros`. However you can specify the path. For example, `ROS_PKG_PREFIX=/opt/ros/indigo hr install hr_msgs`

AmeBel, 2017-08-08:
Resolved by https://github.com/hansonrobotics/hrtool/pull/1 and https://github.com/opencog/ros-behavior-scripting/pull/175

closed 2017-08-08 by AmeBel