hansonrobotics/HEAD #692: Settings page and node issues

vytasrgl, 2017-04-26:
1. Currently checkbox values are not loaded properly, and the settings are updated based on unloaded settings which may lead to udesirable settings changes.
2. On timeline editor only selected properties must be saved/and loaded from file.
3. After visiting settings page JS exception in hompage: layout.js:49 Uncaught TypeError: this.ui.navigation.multilevelpushmenu is not a function

vytasrgl, 2017-04-26:
URLS for nodes doesnt work for nodes too

grebsle, 2017-04-26:
https://github.com/hansonrobotics/HEAD/pull/694

vytasrgl, 2017-05-04:
1,2 still not fixed.

In addition in 2. Wrong current node schema is saved to file.

grebsle, 2017-05-08:
https://github.com/hansonrobotics/HEAD/pull/717

closed 2017-05-09 by vytasrgl