hansonrobotics/HEAD #709: Relex throws exception running OpenCog chatbot/behavior tree

eddiemonroe, 2017-05-04:
On a fresh installation of HEAD, running

./dev.sh --oc

results in java.lang.IndexOutOfBoundsExceptions in the relex server. 

This leads to "The Relex server seems to have crashed!" messages in and aborting of the OpenCog scripts in the "cog" and "tel" windows.

Here is the output from the "relex server" window:

Info: hndlr=1 sentence: I am looking to the left
Error: Failed to process sentence: I am looking to the left
java.lang.IndexOutOfBoundsException: Index: 7, Size: 7
        at java.util.ArrayList.rangeCheck(ArrayList.java:635)
        at java.util.ArrayList.get(ArrayList.java:411)
        at relex.ParsedSentence.getWordAsNode(ParsedSentence.java:167)
        at relex.parser.LocalLGParser.addLinkageStructure(LocalLGParser.java:322)
        at relex.parser.LocalLGParser.parse(LocalLGParser.java:244)
        at relex.RelationExtractor.parseSentence(RelationExtractor.java:308)
        at relex.RelationExtractor.processSentence(RelationExtractor.java:268)
        at relex.ServerSession.handle_session(ServerSession.java:164)
        at relex.Server.run(Server.java:203)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor.run(ThreadPoolExecutor.java:615)
        at java.lang.Thread.run(Thread.java:745)
RelEx processing: 15 milliseconds (avg=15 millisecs, cnt=1)
Error: hndlr=1 Failed to parse: null
java.lang.NullPointerException
        at relex.ServerSession.handle_session(ServerSession.java:165)
        at relex.Server.run(Server.java:203)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor.run(ThreadPoolExecutor.java:615)
        at java.lang.Thread.run(Thread.java:745)
Info: hndlr=1 Closed input socket
Loop count=2 Restart count=0
Info: Waiting for socket connection
Info: Enter thread with handler 2
Info: hndlr=2 recv input: small cats are cute
Info: hndlr=2 sentence: small cats are cute
Error: Failed to process sentence: small cats are cute
java.lang.IndexOutOfBoundsException: Index: 5, Size: 5
        at java.util.ArrayList.rangeCheck(ArrayList.java:635)
        at java.util.ArrayList.get(ArrayList.java:411)
        at relex.ParsedSentence.getWordAsNode(ParsedSentence.java:167)
        at relex.parser.LocalLGParser.addLinkageStructure(LocalLGParser.java:322)
        at relex.parser.LocalLGParser.parse(LocalLGParser.java:244)
        at relex.RelationExtractor.parseSentence(RelationExtractor.java:308)
        at relex.RelationExtractor.processSentence(RelationExtractor.java:268)
        at relex.ServerSession.handle_session(ServerSession.java:164)
        at relex.Server.run(Server.java:203)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor.run(ThreadPoolExecutor.java:615)
        at java.lang.Thread.run(Thread.java:745)
RelEx processing: 5 milliseconds (avg=5 millisecs, cnt=1)
Error: hndlr=2 Failed to parse: null
java.lang.NullPointerException
        at relex.ServerSession.handle_session(ServerSession.java:165)
        at relex.Server.run(Server.java:203)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor.run(ThreadPoolExecutor.java:615)
        at java.lang.Thread.run(Thread.java:745)
Info: hndlr=2 Closed input socket

leungmanhin, 2017-05-04:
The relex server crashes when parsing a sentence, looks like it's related to this: https://github.com/opencog/relex/commit/2c18f5a89cacdfd1e4a174652afd7c8089304270

eddiemonroe, 2017-05-04:
Thanks @leungmanhin, I filed an issue in the relex repo. https://github.com/opencog/relex/issues/256

leungmanhin, 2017-05-11:
I think this is fixed by https://github.com/opencog/relex/commit/301695ce024b7db150009b93d99f92299b30dbbe