hansonrobotics/HEAD #740: Keyboard shortcuts

vytasrgl, 2017-05-29:
Timeline editor (edit mode):
Ctrl + A Select All nodes
Ctrl + C copy selected nodes), visual indication ("N node(s) copied")
Ctrl + V Paste at the current indicator time
Ctrl + S (save) (same visual indication)
Ctrl + D
Del - delete selected nodes
Space or Alt+a (Run/pause)
[ - previous timeline
] - next timeline

Save dialog:
Y - Yes
N - No
Esc (Cancel)

Playback (non-edit mode, also should be in puppeteering mode)
Space OR ALT+a (Run/pause)
[ - previous timeline
] - next timeline

grebsle, 2017-06-01:
https://github.com/hansonrobotics/HEAD/pull/743

vytasrgl, 2017-06-02:
CTRL+D - Should be same as Done 
CTRL+E - Should be same as Edit (for the perfromance page)

vytasrgl, 2017-06-03:
CTRL+P Enable autopause
CTRL+L Enable loop

grebsle, 2017-06-07:
https://github.com/hansonrobotics/HEAD/pull/749

vytasrgl, 2017-06-21:
CTRL+N should be "Add new"

grebsle, 2017-07-25:
https://github.com/hansonrobotics/HEAD/pull/782

closed 2017-07-25 by vytasrgl