hansonrobotics/HEAD #753: Performances trigered by keyword or API

vytasrgl, 2017-06-21:
Sometimes the performance triggered by the speech cant play properly, and or finish because of UI running on browser.
Also performances triggered by events on timeline Pause node are not correctly handled then UI is open, as there seems to be some callbacks from UI that breaks the flow.

grebsle, 2017-07-25:
Unable to reproduce the issue. Need more info

vytasrgl, 2017-07-25:
try start two keyword triggered performances at same time

grebsle, 2017-07-27:
Cannot start another performance (wake up) while the first (go to sleep) is still running no matter if UI is open or closed. Sometimes the performance list starts to slide up and down multiple of times. Is this the issue you noticed?

grebsle, 2017-08-01:
The reason why a new performance cannot be played until the end of current performance is the behavior of wholeshow implementation:

perform() method raises an exception inside of try statement preventing new performance from running:
MachineError: "Can't trigger event perform from state performing!"

vytasrgl, 2017-08-01:
its fine, and its made that way so wholeshow settings remain the same while
switching from perfoming state to performing state, doesnt require to
modify any chatbot settings or existing behavior settings

grebsle, 2017-08-15:
https://github.com/hansonrobotics/HEAD/tree/runner_interrupts

Seems to be working well, but needs more testing and code cleaning.

grebsle, 2017-08-18:
https://github.com/hansonrobotics/HEAD/pull/798

Interrupts seems to be working well for one level nesting, but fails on deeper levels.

closed 2017-08-29 by vytasrgl