hansonrobotics/HEAD #755: Enable/disable timelines in performance

vytasrgl, 2017-06-21:
Enable/Disable timelines in performance. 
On Performance queue allow enable/disable specific timelines.
Disabled timelines should not be rendered on the merged timeline, so after the change the view would need to be updated.
"enabled" property is optional in the timeline file and if its not present it means timeline is enabled. That ensures backward compatibility.

This will be exception for puppeteering view as enabled/disabled should be working on the puppeteering view because it may be useful to have it on run-time. Enable/Disabled should not be allowed while running the performance.

grebsle, 2017-08-27:
https://github.com/hansonrobotics/HEAD/pull/802

vytasrgl, 2017-08-28:
it takes 2-3secs to update the view. Must be loading indicator for such actions

grebsle, 2017-08-30:
https://github.com/hansonrobotics/HEAD/pull/812