hansonrobotics/HEAD #831: How do I actually speak with the HEAD robot?

deviousname, 2018-01-17:
I open the browser to 127.0.0.1:8000 and I can see the Hanson robotics menus, I can see the HEAD in blender, but when I type something in the chat, I get nothing in response. Am I missing something?

KineticIsEpic, 2018-01-18:
Personally I get this error during start-up:
```
Traceback (most recent call last):
  File "/home/developer/hansonrobotics/HEAD/src/ros_tts/scripts/tts_talker.py", line 454, in <module>
    talker = TTSTalker()
  File "/home/developer/hansonrobotics/HEAD/src/ros_tts/scripts/tts_talker.py", line 53, in __init__
    self.mongodb = MongoDB()
TypeError: __init__() takes exactly 2 arguments (1 given)
[robot/chatbot-7] process has died [pid 30995, exit code 1, cmd /home/developer/hansonrobotics/HEAD/src/chatbot/scripts/ai.py /home/developer/hansonrobotics/HEAD/src/chatbot/aiml/ -sent __name:=chatbot __log:=/home/developer/.ros/log/943cdcb2-fbec-11e7-9ab6-100ba935d678/robot-chatbot-7.log].
log file: /home/developer/.ros/log/943cdcb2-fbec-11e7-9ab6-100ba935d678/robot-chatbot-7*.log
```
Which prevents the script from running. Granted, I installed slightly differently ("new" hrtool in a Docker container), which brought up its own slew of issues. Check your terminal when you launch the package and see if you get an error similar to mine, if so there's probably a code issue.

wenwei-dev, 2018-01-24:
@KineticIsEpic This bug should have been fixed. Please update chatbot repo to the latest master branch and try again.