hansonrobotics/HEAD #832: webui broken dependencies

tim777z, 2018-02-05:
running: ~/hansonrobotics/HEAD/scripts/hrtool -B
returns
npm ERR! Error: No compatible version found: supermodel@'github:pathable/supermodel'
npm ERR! Valid install targets:
npm ERR! ["0.0.1"]
npm ERR!     at installTargetsError (/usr/share/npm/lib/cache.js:719:10)
npm ERR!     at /usr/share/npm/lib/cache.js:638:10
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:142:7)
npm ERR!     at /usr/lib/nodejs/graceful-fs/polyfills.js:133:7
npm ERR!     at Object.oncomplete (fs.js:107:15)
npm ERR! If you need help, you may report this log at:
npm ERR!     <http://github.com/isaacs/npm/issues>
npm ERR! or email it to:
npm ERR!     <npm-@googlegroups.com>

npm ERR! System Linux 4.13.0-32-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /root/hansonrobotics/HEAD/src/webui
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.3.10

any sugesstions?

elggem, 2018-03-15:
Try updating npm/node to newer versions, there have been issues under 14.04 with older versions.

tim777z, 2018-03-15:
Thank you, Yes, thats the best solution, but I'm having another issue as well, I will close this issue and open another one

closed 2018-03-15 by tim777z